uhi22/ccs32clara #6: Question: Magic Numbers PEV state machine

f0ld, 2023-12-18:
Hi,

Is there a deeper meaning behind the tokens/numbers defined at the top of pevStateMachine.cpp for the different states? Are they just to simplify the code or do they have a distinct meaning/origin (e.g. in IEC Standard) ?

Thanks in advance for the answer.

uhi22, 2023-12-18:
You are referring to the definitions of PEV_STATE_xyz? A lot of these states are indeed given by the standard (only know the ISO15118, but the DIN is not far away from this.) The first specification-based state is PEV_STATE_WaitForSupportedApplicationProtocolResponse, the last is PEV_STATE_WaitForSessionStopResponse. Outside of this "official" range we created some extra states, like PEV_STATE_Connecting or PEV_STATE_SafeShutDownWaitForChargerShutdown. These are not directly derived from the spec, but come from practical experiences.
The number of each of the state does not matter at all, it is just "random".

f0ld, 2023-12-18:
> You are referring to the definitions of PEV_STATE_xyz? A lot of these states are indeed given by the standard (only know the ISO15118, but the DIN 70121 is not far away from this.) The first specification-based state is PEV_STATE_WaitForSupportedApplicationProtocolResponse, the last is PEV_STATE_WaitForSessionStopResponse. Outside of this "official" range we created some extra states, like PEV_STATE_Connecting or PEV_STATE_SafeShutDownWaitForChargerShutdown. These are not directly derived from the spec, but come from practical experiences. The number of each of the state does not matter at all, it is just "random".

Yes. That part that the states are derived from the standard I knew. But I was confused by the numbers used as a aliases (suspected they are "random"). Didnt want to do something wrong when I add states.

So this solves my question. 
Thank you for the quick reply 👍

closed 2023-12-18 by f0ld