uhi22/ccs32clara #13: ide

Fatih0, 2024-02-11:
Can we use cubeide with this code? How can we use this code in any ide. Can you recommend ide?
if we use cubeide, wiil be perfect

f0ld, 2024-02-11:
You can find the manual here:
https://github.com/uhi22/ccs32clara/blob/main/doc/clara_user_manual.md

It currently does not use cubeide (although you can). You can use whatever IDE you feel comfortable with since a makefile is provided.

Or as stated in the documentation as well there is always more info in the supporting forum thread on OpenInverter (where you can also find out more about the history of decisions made).
There is also what you are probably looking for here in this thread:

https://openinverter.org/forum/viewtopic.php?t=3727&start=325

Fatih0, 2024-02-11:
Can I enter the debug session with cubeide? libopencm3 is different so I dont know how is impelement
thanks

jsphuebner, 2024-02-12:
Did you try it?

f0ld, 2024-02-13:
> Can I enter the debug session with cubeide? libopencm3 is different so I dont know how is impelement thanks

Do you actually have the FOCCI Hardware? Clara is meant to be used on FOCCI or similar hardware. 

Are you talking about Software or Hardware Debugging? 

I dont want to discourage you (quite the opposit actually) but without even simple information (e.g. what it is in general you are trying to do), it is very hard to help you. This repository is also still heavy work in progress and doesnt claim completeness in any way or form. That means yes collaboration is very welcome but certain things/paths might not be explored (yet). 
So in general my recommendation is to read the accompanying thread on the OpenInverter as well as the documentation already provided. 

Regarding debugging:
Clara talks (by design) a lot over the serial connection for logging and troubleshooting. So if you intend to do HW debugging I dont see any sense in doing troubleshooting that way. Code errors will be catched by the compiler. A lot of the logic errors will show itself when running/testing the code on the hardware via SerialPort (especially since you cant catch some of those during regular debugging sessions). From a resource standpoint there is also no point in using advanced debugging features as Focci and Clara are far away from that being a limiting factor...

That doesnt meen you cant try it. It technically should work but to my knowledge nobody has tried it with the current stack as this is not the best/most convenient way to do things at the moment. 

If you try it out feedback regarding this is highly welcomed!

Fatih0, 2024-02-15:
I have tested with pyPLC on raspberrypi normally. Also I have codico hardware. I guess, I tested ccsclara with stm32 6 months ago. You pass the libopen librarry, I didnt use it before. I asked my questions. now Im not using. best wishes. before I charged over dc charging stations, also I have implemented ac charge session.

uhi22, 2024-02-15:
Coming back to your question "Can I enter the debug session with cubeide?", I summarize what I interpret from the discussiones above, in short words: "Nobody tried until now."

My personal longer answer: With changing from the cubeide project to the makefile approach the "intrinsic" CubeIDE support is removed. This makes it harder for somebody, who is used to the CubeIDE. For my daily work it is no loss, I'm fine with notepad++ and the make on command line, and the serial logging. Nevertheless it is a senseful goal to make the CubeIDE running again. So you (and everybody else) is invited to find out what is missing, what are the obstacles and what are solutions. My stomage feeling is, that it should be possible.
So finally an other answer to your question "Can I enter the debug session with cubeide?": Yes, sure, if you find out the obstacles and find out solutions, you can. In best case, share your knowledge with the community. Thanks for this.

jsphuebner, 2024-02-15:
CubeIDE was also discussed here: https://openinverter.org/forum/viewtopic.php?p=65181#p65181
Memeber olegiv posted Eclipse/CubeIDE project files a few posts further down

closed 2024-05-25 by jsphuebner