uhi22/ccs32clara #15: incoming out-of-interest TCP data overwrites the before correctly received TCP data

uhi22, 2024-02-18:
In the case, that the charger sends a valid TCP message and afterwards an other TCP message on a port which Clara is not interested in, the later data overwrites the correctly received message, so that the state machine does not see it. Clara runs into timeout due to this.
Observed here:
https://[raw.githubusercontent.com/uhi22/clara-logs/main/2024-02-18_clara_TeslaV3_lightbulb_ok_at_second_try.claralog.decoded.txt](https://raw.githubusercontent.com/uhi22/clara-logs/main/2024-02-18_clara_TeslaV3_lightbulb_ok_at_second_try.claralog.decoded.txt)
* At 18650ms, the charger sends a ChargeParameterDiscoveryResponse.
* At 18770ms, the charger sends a TCP message on an other port (reason unknown, but does not matter).
* At 19520ms, the state machine looks on the received message (state machine was in the 1s-do-nothing-phase). It does not see a valid response, because the response was overwritten.

jsphuebner, 2024-02-18:
Is this related to #10 ?

uhi22, 2024-02-18:
This is not clear. There was not tracing of the ethernet in this case, so we can only guess.

Analysis: In evaluateTcpPacket(), there is already the check for correct port, and the function exits if the ports do not match. So the idea would be, to NOT write data to tcp_rxdata if the ports are wrong. But with the approach, that tcp_rxdata always points to myethreceivebuffer[74], new incoming data, even if they are not related to tcp or have the wrong port, are making that the old, valid data is overwritten before use.

jsphuebner, 2024-02-18:
Ok that would be easy to undo

jsphuebner, 2024-02-18:
Wait other question: since data is polled from the QCA chip in the 30ms task how can it not be processed?

uhi22, 2024-02-18:
The state machine, after sending the chargeParameterDiscoveryRequest, makes a one-second break, until it looks to the response. The processing happens, if the one second is elapsed. So if something comes in between, the state machine does not see the response. Yes, this "waiting while the response is already there" is also not the best design. It would be better, to react on the response immediately, and add the wait state afterwards. The wait time itself is necessary to avoid that each 30ms a ping-pong happens, while the charger anyway needs one or more seconds to find its charger parameters.

uhi22, 2024-02-19:
The wiresharks view:
[image]

closed 2024-02-20 by uhi22