uhi22/ccs32clara #19: STM32F103RE Is there enough RAM space?

IamTomChang, 2024-05-24:
The data structure space of DIN EXI is about 40K, but the ram space of STM32F103RE is only 64K ram, is this enough, and is there any good solution?

jsphuebner, 2024-05-25:
Well 40k < 64k and the other variables only need about 15k and we don't expect much growth. So not really an issue

uhi22, 2024-05-26:
What means "good solution"? For which problem?

IamTomChang, 2024-05-27:
Indeed, as you said, 40k < 64K, which is enough for the DIN protocol. But what I mean by this is that if I want it to support both DIN and ISO protocols, then 64K ram space is not enough, so is there a better solution for this than to replace the MCU with more RAM space?

uhi22, 2024-05-27:
The combination of DIN and ISO1 (aka ISO 2013) seems to fit well, see https://openinverter.org/forum/viewtopic.php?p=70756#p70756
If your goal is, to use the ISO2 (aka ISO 2016) with the certificates for plug'n'charge, then a bigger controller is necessary.

By the way: What is the motivation, to jump from DIN to ISO? Which features would you get with ISO, that are missing in the DIN? Did you find a kind of difference description between the DIN, ISO 2013 and ISO 2016 (and younger)? Would be real helpful to understand where we are and where we could go to. Thanks.

IamTomChang, 2024-05-28:
Because I am also new to CCS charging technology, as you said, DIN is the first version of CCS DC charging protocol, which can basically meet the general charging needs. However, the current market demand is still very keen on TLS, PnC, AC advanced communication and bidirectional charging technology, these functions are not supported by DIN, and ISO15118 protocol is supported. This is the reason I can think of to switch from DIN to ISO.

IamTomChang, 2024-05-28:
I have two more questions, and I'd like you to see if you can answer them?
1.
We know that when we use AVitar software to burn firmware to QCA7000, the MAC configuration option of QCA7000 will appear, I would like to ask what is the function of this MAC? When communicating HLC with the Host MCU, must the MAC address in the Ethernet frame be the same as the MAC address when the firmware was burned? We have tested that the Host MCU uses other MAC addresses, and the communication with other modules can also be normal. Is it unclear what the specific role of the MAC of the QCA7000 is?
2.
When SLAC is in the Linked_ok state, the cpu_on status bit in INTR_CAUSE is set？But we found in the data book that only when the chip is powered on will the interrupt be triggered and the bit cpu_on be set？Is this a normal phenomenon?Now we suspect that it is SET_KEY_Req, QCA7005 a reset action will occur when joining the network (the reset process will last about 1s), but we have not found relevant information in the data sheet, so we don't know if our guess is correct?

uhi22, 2024-06-25:
Closing, because original topic is discussed, and for further topics better use the forum (openinverter.org)

closed 2024-06-25 by uhi22