uhi22/ccs32clara #22: Sequential contactor control

jsphuebner, 2024-06-26:
Right now we enable both contactor outputs simultaneously at 100% duty cycle when turning on.

From a power management point of view it would make more sense to enable them sequentially. Also the 100% turn on pulse could be reduced, by my experience, to 50-100ms.
So the turn on sequence would be
1. Turn on CONT1 @ 100%
2. Reduce CONT1 to EconomizerDuty
3. Turn on CONT2 @ 100%
4. Reduce CONT2 to EconomizerDuty

uhi22, 2024-06-26:
Done:
[image]

closed 2024-06-26 by uhi22

uhi22, 2024-08-07:
We found an unwanted side effect of this sequential control, discussed here: https://openinverter.org/forum/viewtopic.php?p=73640#p73640 : If the charger is fastly providing current, it finds the second contactor still open, and runs into error because of missing battery connection. This was found on a Kempower.
The needed solution is to delay the currentDemand, so that it starts only after both contactors are closed.

uhi22, 2024-08-07:
Improvement added here: https://github.com/uhi22/ccs32clara/commit/5b543714b59a24d1be31e837c9edf6f621c33ad9

commit 17fb48a8 (2024-06-26): feature: sequential contactor control to reduce current peak. Fixes #22.