uhi22/ccs32clara #24: Neighbor CCS session stops the own session

uhi22, 2024-07-13:
Precondition: Clara is running a charging session e.g. on an Alpitronics charger.
Action: On the second CCS of the same charger, plug-in an other EV and authorize
Observed: Claras session stops
Expected: Clara continues charging

Observed by Johu with the Touran, and also here: https://openinverter.org/forum/viewtopic.php?p=72652#p72652

Reproducable on the desk.

The clara log shows
`
[87600] Data received:  01fe80010000001e809802004080c1014181c210e00000002040c40de01030c0042006021fe8
[87720] Data received:  01fe80010000001e809802004080c1014181c210e00000002040c40de01030c0042006021fe8
[87780] [CONNMGR] 165 0 0 0 0 162 328 --> 100
[87810] Data received:  01fe80010000001e809802004080c1014181c210e00000002040c40de01030c0042006021fe8
[87840] [PEVSLAC] received SLAC_MATCH.CNF
[87840] [PEVSLAC] From SlacMatchCnf, got network membership key (NMK) and NID.
[87840] [PEVSLAC] Checkpoint170: transmitting CM_SET_KEY.REQ
[87840] ETH will transmit: ffffffffffff02353069204388e1010860000001aaaaaaaa0000000004000000000102812c4f1307017777244e81ecf0777777777777777777000000
[87840] ETH rx IP: 023530692043b827ebe6398986dd60013c7600140640fe800000000000005f92a4a1f5e8a025fe80000000000000c69083f3fbcb981e3b0ee755ebaa7563000072b2501001fa42030000
[87870] ETH rx HP: 023530692043046565ffffff88e10109600000016b4fb694aaaaaaaa040000ff00000000000000000000000000000000000000000000000000000000
[87870] [PEVSLAC] received SET_KEY.CNF
[87870] [PEVSLAC] SetKeyCnf says 1, this is formally 'rejected', but indeed ok.
`

This means, while in the CurrentDemand loop, Clara gets a SLAC_MATCH.CNF (from the other charger), and uses the network management key from the other session. Clara configured the modem with the new key, and so joins the neighbors network.
In an other test run on the desk, joining the neighbors network can be seen at the SDP, which Clara never requested:
`
[66480] Data received:  01fe80010000001e809802004080c1014181c210e00000002040c40de01030c0042006021fe8
[66570] Data received:  01fe80010000001e809802004080c1014181c210e00000002040c40e001030c0042006021fe8
[66660] Data received:  01fe80010000001e809802004080c1014181c210e00000002040c40de01030c0042006021fe8
[66720] [SDP] Checkpoint203: Received SDP response
`

uhi22, 2024-07-13:
Solution Concept: While Clara has a valid TCP connection (ConnectionLevel=100), it shall ignore all SLAC and SDP traffic.

uhi22, 2024-07-13:
With the fix, now the log shows while we are in the charging loop:

```
[176010] Data received:  01fe80010000001c809a02004080c1014181c210e000408000018286f8081818008400c0
[176010] TCP will transmit: 01fe800100000023809a02004080c1014181c210d1400000c0c0286001810580480c08360100c143780400
[176100] [TCP] sending ACK
[176100] Data received:  01fe80010000001c809a02004080c1014181c210e000408000018286f8081818008400c0
[176100] [HOMEPLUG] Ignoring homeplug message, because high level communication is ongoing.
[176100] TCP will transmit: 01fe800100000023809a02004080c1014181c210d1400000c0c0286001810580480c08360100c143780400
[176130] In state CurrentDemand. TcpRetries 0
[176190] [TCP] sending ACK
[176190] Data received:  01fe80010000001c809a02004080c1014181c210e000408000018286f0081818008400c0
[176190] TCP will transmit: 01fe800100000023809a02004080c1014181c210d1400000c0c0286001810580480c08360100c143780400
[176220] [CONNMGR] 165 0 0 0 0 163 329 --> 100
[176310] [TCP] sending ACK
[176310] Data received:  01fe80010000001c809a02004080c1014181c210e000408000018286f0081818008400c0
[176310] TCP will transmit: 01fe800100000023809a02004080c1014181c210d1400000c0c0286001810580480c08360100c143780400
[176400] [TCP] sending ACK
[176400] Data received:  01fe80010000001c809a02004080c1014181c210e000408000018286f8081818008400c0
[176400] TCP will transmit: 01fe800100000023809a02004080c1014181c210d1400000c0c0286001810580480c08360100c143780400
[176490] [TCP] sending ACK
[176490] Data received:  01fe80010000001c809a02004080c1014181c210e000408000018286f8081818008400c0
[176490] TCP will transmit: 01fe800100000023809a02004080c1014181c210d1400000c0c0286001810580480c08360100c143780400
[176580] [TCP] sending ACK
[176580] Data received:  01fe80010000001c809a02004080c1014181c210e00040800001828700081818008400c0
[176580] TCP will transmit: 01fe800100000023809a02004080c1014181c210d1400000c0c0286001810580480c08360100c143780400
[176670] [TCP] sending ACK
[176670] Data received:  01fe80010000001c809a02004080c1014181c210e00040800001828700081818008400c0
[176670] TCP will transmit: 01fe800100000023809a02004080c1014181c210d1400000c0c0286001810580480c08360100c143780400
[176700] [SDP] Ignoring SDP response, because high level communication is ongoing.
[176700] [HOMEPLUG] Ignoring homeplug message, because high level communication is ongoing.
[176730] [TCP] sending ACK
[176730] Data received:  01fe80010000001c809a02004080c1014181c210e000408000018286f8081818008400c0
[176730] TCP will transmit: 01fe800100000023809a02004080c1014181c210d1400000c0c0286001810580480c08360100c143780400
```
It is still unclear, why the foreign SDP response is visible at all. This looks still like a mix-up of the two AVLNs. To be further analyzed.

jsphuebner, 2024-07-18:
Will update and test with a friend on a nearby Aral Pulse (HyperCharger)

jsphuebner, 2024-07-24:
Tested on Aral Pulse today:
- Plugged in and charged at 70 kW
- another car arrived and kindly agreed to use the other port (instead of another charger)
- When they started my charge power briefly dropped to 50 kW but after a few seconds resumed to 70 kW
- The session was NOT interrupted!

uhi22, 2024-08-20:
Test was successful and no more complaints -> close

closed 2024-08-20 by uhi22