uhi22/ccs32clara #25: Dependencies changed libopencm3 causes issues

Tom-evnut, 2024-07-18:
I tried first downloading code via Github Desktop and would not complete due to issues with the libopencm3 and libopeninv reference.

Copied over the libopencm3 and libopeninv as referenced from github and it fails:

```/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: obj/hwinit.o: in function `clock_setup':
/app/ccs32clara/src/hwinit.cpp:43: undefined reference to `rcc_clock_setup_pll'
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: /app/ccs32clara/src/hwinit.cpp:65: undefined reference to `rcc_hse_configs'
collect2: error: ld returned 1 exit status
make: *** [Makefile:113: stm32_ccs] Error 1
Press any key to continue . . .```

Tom-evnut, 2024-07-18:
```
  GIT SUBMODULE
fatal: could not get a repository handle for submodule 'libopencm3'
make: *** [Makefile:154: get-deps] Error 1
```

jsphuebner, 2024-07-18:
That is really strange, just did a fresh clone and was able to build

```
> git clone git@github.com:uhi22/ccs32clara.git
Klone nach 'ccs32clara'...
remote: Enumerating objects: 1781, done.
remote: Counting objects: 100% (404/404), done.
remote: Compressing objects: 100% (154/154), done.
remote: Total 1781 (delta 260), reused 381 (delta 246), pack-reused 1377
Empfange Objekte: 100% (1781/1781), 20.24 MiB | 520.00 KiB/s, fertig.
Löse Unterschiede auf: 100% (1140/1140), fertig.

> make get-deps
  GIT SUBMODULE
Submodul 'libopencm3' (git@github.com:jsphuebner/libopencm3.git) für Pfad 'libopencm3' in die Konfiguration eingetragen.
Submodul 'libopeninv' (git@github.com:jsphuebner/libopeninv.git) für Pfad 'libopeninv' in die Konfiguration eingetragen.
Klone nach '/tmp/ccs32clara/libopencm3'...
Klone nach '/tmp/ccs32clara/libopeninv'...
Submodul-Pfad 'libopencm3': '3413ef88067558cba8be90be6aa339a33788f88e' ausgecheckt
Submodul-Pfad 'libopeninv': '613097e26ae578b032c320970be47d3428232fbb' ausgecheckt
  MAKE libopencm3
  GENHDR  stm32/f1
  BUILD   lib/stm32/f1
  CC      adc.c
  CC      adc_common_v1.c
  CC      can.c
  CC      crc_common_all.c
  CC      dac_common_all.c
  CC      dac_common_v1.c
  CC      desig_common_all.c
  CC      desig_common_v1.c
  CC      dma_common_l1f013.c
  CC      exti_common_all.c
  CC      flash.c
  CC      flash_common_all.c
  CC      flash_common_f.c
  CC      flash_common_f01.c
  CC      gpio.c
  CC      gpio_common_all.c
  CC      i2c_common_v1.c
  CC      iwdg_common_all.c
  CC      pwr_common_v1.c
  CC      rcc.c
  CC      rcc_common_all.c
  CC      rtc.c
  CC      spi_common_all.c
  CC      spi_common_v1.c
  CC      timer.c
  CC      timer_common_all.c
  CC      usart_common_all.c
  CC      usart_common_f124.c
  CC      mac.c
  CC      mac_stm32fxx7.c
  CC      phy.c
  CC      phy_ksz80x1.c
  CC      usb.c
  CC      usb_control.c
  CC      usb_standard.c
  CC      usb_msc.c
  CC      usb_hid.c
  CC      usb_audio.c
  CC      usb_cdc.c
  CC      usb_midi.c
  CC      usb_dwc_common.c
  CC      usb_f107.c
  CC      st_usbfs_core.c
  CC      st_usbfs_v1.c
  CC      vector.c
  CC      systick.c
  CC      scb.c
  CC      nvic.c
  CC      assert.c
  CC      sync.c
  CC      dwt.c
  AR      libopencm3_stm32f1.a

> make
  CPP     obj/main.o
  CPP     obj/hwinit.o
  CPP     obj/stm32scheduler.o
  CPP     obj/params.o
  CPP     obj/terminal.o
  CPP     obj/terminal_prj.o
  CC      obj/my_string.o
  CPP     obj/digio.o
  CC      obj/my_fp.o
  CPP     obj/printf.o
  CPP     obj/anain.o
  CPP     obj/param_save.o
  CPP     obj/errormessage.o
  CPP     obj/stm32_can.o
  CPP     obj/canhardware.o
  CPP     obj/canmap.o
  CPP     obj/cansdo.o
  CPP     obj/picontroller.o
  CPP     obj/terminalcommands.o
  CPP     obj/ipv6.o
  CPP     obj/tcp.o
  CPP     obj/connMgr.o
  CPP     obj/modemFinder.o
  CPP     obj/pevStateMachine.o
  CPP     obj/temperatures.o
  CPP     obj/proximitypilot.o
  CPP     obj/acOBC.o
  CPP     obj/hardwareInterface.o
  CPP     obj/hardwareVariants.o
  CPP     obj/pushbutton.o
  CPP     obj/udpChecksum.o
  CPP     obj/homeplug.o
  CPP     obj/myHelpers.o
  CPP     obj/qca7000.o
  CC      obj/appHandEXIDatatypesDecoder.o
  CC      obj/ByteStream.o
  CC      obj/EncoderChannel.o
  CC      obj/appHandEXIDatatypesEncoder.o
  CC      obj/DecoderChannel.o
  CC      obj/EXIHeaderDecoder.o
  CC      obj/appHandEXIDatatypes.o
  CC      obj/dinEXIDatatypesDecoder.o
  CC      obj/EXIHeaderEncoder.o
  CC      obj/BitInputStream.o
  CC      obj/dinEXIDatatypesEncoder.o
  CC      obj/MethodsBag.o
  CC      obj/BitOutputStream.o
  CC      obj/dinEXIDatatypes.o
  CC      obj/projectExiConnector.o
  LD      stm32_ccs
  OBJCOPY stm32_ccs.bin
  OBJCOPY stm32_ccs.hex
   text	   data	    bss	    dec	    hex	filename
 141392	   2208	  57560	 201160	  311c8	stm32_ccs
```

Tom-evnut, 2024-07-18:
I wiped the code and tried pulling it again with the [zombuild scripts](https://github.com/crasbe/ZombieBuild) scripts pointed towards this repository and this is what it fails on. Must be the same reason the Github Desktop app fails. I do not use Github enough to fully understand it

```
 GIT SUBMODULE
Submodule 'libopencm3' (git@github.com:jsphuebner/libopencm3.git) registered for path 'libopencm3'
Submodule 'libopeninv' (git@github.com:jsphuebner/libopeninv.git) registered for path 'libopeninv'
Cloning into '/app/ccs32clara/libopencm3'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
fatal: clone of 'git@github.com:jsphuebner/libopencm3.git' into submodule path '/app/ccs32clara/libopencm3' failed
Failed to clone 'libopencm3'. Retry scheduled
Cloning into '/app/ccs32clara/libopeninv'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
fatal: clone of 'git@github.com:jsphuebner/libopeninv.git' into submodule path '/app/ccs32clara/libopeninv' failed
Failed to clone 'libopeninv'. Retry scheduled
Cloning into '/app/ccs32clara/libopencm3'...
error: cannot run ssh: No such file or directory
fatal: unable to fork
fatal: clone of 'git@github.com:jsphuebner/libopencm3.git' into submodule path '/app/ccs32clara/libopencm3' failed
Failed to clone 'libopencm3' a second time, aborting
```

Tom-evnut, 2024-07-18:
Fix found by CrasBe

https://github.com/uhi22/ccs32clara/pull/26

closed 2024-07-18 by Tom-evnut