uhi22/ccs32clara #38: Add some BMS functionality

uhi22, 2024-09-15:
Normally, the target values for voltage and current are calculated from the BMS.
But there seem to be use cases, where the installed BMS only provides "low level" data like battery voltage and module temperatures.
For these case, Foccci/Clara could provide some basic high-level-BMS functionality, e.g.
- calculate the target charge current based on the pack voltage (like P.S.Mangelsdorf does as quick-and-dirty approach)
- calculate the target charge current based on the min and max cell voltage
- calculate the target charge current based on the SOC
- estimate an SOC based on the pack voltage, to be able to show something at the chargers display
- derating of the target charge current in case of low cell temperature and high cell temperature
- ...maybe some more....

This issue is intended to collect the ideas first, and make a concept how all the different possibilities and requirements can fit together.

Tom-evnut, 2024-09-15:
I would avoid this, it should remain a "dumb gateway" in my opinion. These are clear tasks for a BMS and should be handled as such. It is a massive scope creep and opens up a ton of ways to create problems.

Or at least have an easy way of suppressing all of this.

uhi22, 2024-09-15:
Yes, the idea would be to have parameters like "dirtybms_calcCurrentFromPackVoltage yes/no" with default "No".

jsphuebner, 2024-09-30:
I'm also afraid we might open a can of worms with this.
If anything I'd limit it to the absolute basics. Charge current based on maximum cell voltage, minimum and maximum temperature. We might even enable it with a compile time switch so we don't taint the "vanilla" release with this hacky approach. Then the build action can create two binaries and you have to explicitly flash the hacky one.

jsphuebner, 2024-10-23:
I talked to Alex today and he found the CV mode of chargers works sufficiently well. I.e. his BMS makes sure all cells are balanced and consequently it is sufficient to set the voltage setpoint of the charger to Number_of_cells x Max_voltage_per_cell.

If anything goes wrong the BMS will stop the charging process with the enable flag.
So the only person that requested the feature now no longer needs it. Close?

uhi22, 2024-10-24:
Yes lets close. If somebody wants to continue we can re-open.

closed 2024-10-24 by uhi22