uhi22/ccs32clara #46: During AC charging lock is not operated correctly if it has no feedback

jsphuebner, 2024-11-22:
In the newly introduced StandAlone AC charge mode the lock misbehaves:
- We do not get past https://github.com/uhi22/ccs32clara/blob/612437f2d668f451a9fc125cd05e32fd9b52c8ee/ccs/acOBC.cpp#L338 because initial LockState is None not Open
- when manipulating (see below) LockOpenThresh the lock closes but the lock motor keeps running, ignoring LockRunTime
- When reverting LockOpenThresh the motor stops and charging starts

Parameters:
- LockRunTime 1500
- LockClosedThresh 100
- LockOpenThresh 100

For getting the lock to close LockOpenThresh set to 4000. For stopping the lock motor LockOpenThresh reverted to 100

Tom-evnut, 2024-11-22:
Would it be an idea to force the lock into a known state, open, upon power on?

Tom-evnut, 2024-11-22:
Oh okay it is because you set things to not toggle. Yes do a forced drive to open. Then evaluate if it should be open or closed.

Does you lock not run with feedback? Is it a pure solenoid one?

jsphuebner, 2024-11-22:
It's a customers. His lock sometimes gets stuck so he operates without feedback to at least be able to charge. At the moment of testing the feedback was also disconnected but will be reconnected. Still it should be able to operate even without feedback, just with LockRunTime

jsphuebner, 2024-11-27:
I have tested AC standalone mode with and without feedback and also Actuatortest. It works as expected. Only if CP PWM goes away stateC will stay active. Could be fixed but more of a cosmetic issue I think