uhi22/ccs32clara #58: TCP

Shiyaoy-Smile, 2025-07-16:
The TCP three-way handshake failed. The charging pile received the ACK, but it still keeps sending SYN-ACK repeatedly.

uhi22, 2025-07-16:
A pcap could help to analyze the topic. Which OS is the charging pile running?

Shiyaoy-Smile, 2025-07-16:
> A pcap could help to analyze the topic. Which OS is the charging pile running?

I don't know what operating system this charging pile uses, but it can charge other vehicles. Our self-developed EVCC has this problem now. SLAC and SDP are both normal. This issue occurs during the TCP handshake.

Shiyaoy-Smile, 2025-07-16:
> How do you see that the charging pile receives the ACK? Can you run wireshark or tcpdump on the pile? Do you have a log from Clara?

Yes, I tested the EVCC with the charging pile end simulator. The charging pile end simulator can use Wireshark to capture packets and it shows that the ACK was received. I'll send you the picture.

[image]

jsphuebner, 2025-07-16:
Please share you code on github, otherwise it becomes a cumbersome experience to help out.

Shiyaoy-Smile, 2025-07-17:
> Please share you code on github, otherwise it becomes a cumbersome experience to help out.

The code we are currently using at EVCC is ccs32clara, with almost no modifications. During the testing process, we only replaced the port number, MAC address, and IP address for testing purposes.

uhi22, 2025-07-17:
This sounds like "almost nothing but a lot". Also the message length. Nobody knows what else. You may want to start from a clean basis and check which of the changes causes the SECC to send the retransmission.

Shiyaoy-Smile, 2025-07-17:
> This sounds like "almost nothing but a lot". Also the message length. Nobody knows what else. You may want to start from a clean basis and check which of the changes causes the SECC to send the retransmission.

I've reverted the code to be the same as that of ccs32clara, but the TCP connection failure issue still persists. Since I took more photos, I can't transfer the Wireshark data to my personal device. Please take a look. Thank you.

[image]
[image]
[image]
[image]
[image]
[image]

osexpert, 2025-08-05:
I am experimenting with something that may be related with charger 'Plugit HUBE S': https://github.com/osexpert/ccs32clara-chademo/tree/debug/tcp_fin_experiment

All other chargers has no problem with being closed with a RST. But this changer...refuses. So then I saw no other way than to send/support FIN, but in the easiest and most hackish way possible:-)
Also added support for "Detect TCP keepalive probe (zero payload, ACK flag set, sequence number == TcpAckNr - 1)" that may help with hypercharger 150 randomly killing the connection with RST/FIN. Both experiments are not confirmed.

Thanks.

osexpert, 2025-08-12:
So I am guessing charger is confused from the previous session and somehow has lingering state related to the MAC? (and I assume a new client port is used). Everyone is breaking tcp in their own ways, it seems.

I added the RST to at least try to do the right thing, but maybe just ghosting the connection was actually the best.

ax-iv, 2025-08-12:
In my case, the charging station gave an error specifically for an improperly terminated TCP session. The dump of the exchange confirms this: the station sends a FIN+ACK, does not wait for a response (at least an ACK), and sends another FIN+ACK, after which it gives an error for the TCP session.

ax-iv, 2025-08-12:
A TCP-RST packet is not a normal session termination for either party. For example: https://www.baeldung.com/cs/tcp-ip-reset-flag. The fact is that the charging station is waiting for the session to end normally, but Clara is already sending Slac packets, even with a new port. I will take dumps without TCP-RST, with TCP-RST, and with TCP-FIN+ACK tomorrow and attach them. Yes, it is better to use a proper TCP session termination. Alternatively, you can use the lvip library

osexpert, 2025-08-13:
I created an experimental tcp fin support for clara here: https://github.com/osexpert/ccs32clara-chademo/commits/clara_main_tcp_fin_experiment/
Currently untested, but it least it show my general idea. I will try it when I get time, It may even work:-)

ax-iv, 2025-08-20:
The charging station will not send you anything after the tcp_fin, except for a repeat (dup) of the tcp_fin. By this time, clara has already sent her tcp_fin

osexpert, 2025-08-20:
I don't think we disagree, probably just different code with same function. At least here is how my tcp.cpp looks like, in my fork: https://github.com/osexpert/ccs32clara-chademo/blob/main/ccs/tcp.cpp

osexpert, 2025-08-22:
> Tell me where to increase the timeout, I can't find it exactly. It's rare, but sometimes all 10 "sounds" don't have time to be sent. Usually, only 6 of them are sent in these cases

There are many different timeouts in homeplug.cpp. Do you have a log to show the problem?

osexpert, 2025-08-22:
I see there is a todo here, to increase to 1000ms. You can try and see it this fixes the problem (change from 6 to 10):
```
   myethtransmitbuffer[22]=6; // timeout N*100ms. Normally 6, means in 600ms all sounds must have been tranmitted.
   // Todo: As long we are a little bit slow, lets give 1000ms instead of 600, so that the
   // charger is able to catch it all.
```